RobTillaart/PCF8575 #14: Array of PCF8575 devices?

ear9mrn, 2021-12-12:
I would like to declare an array of multiple PCF8575 devices within a struct, something like:

PCF8575 PCF[DEVICES];

Then, in another part of my code do something like:

PCF[0].begin(sda,scl);
PCF[0].setAddress(address);
PCF[0].read(0);

I cannot figure out how to do this with this library as it appears at minimum I need to assign an address with the constructor.

Any suggestions would be greatly appreciated.

Thanks.

RobTillaart, 2021-12-13:
Thanks for the question.
Never tried this.first thought:
Have you tried to assign a dummy address in the constructor?

RobTillaart, 2021-12-13:
@ear9mrn 

I have no test setup, could you try this code?
```cpp
//
//    FILE: PCF8575_array.ino
//  AUTHOR: Rob Tillaart
//    DATE: 2021-12-13
// PUPROSE: demo array of PCF - not tested
//

#include "PCF8575.h"

// adjust addresses if needed, can be overruled later in setup()
PCF8575 A(0x38);
PCF8575 B(0x39);
PCF8575 C(0x3A);

PCF8575 PCF[3] = { A, B, C };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("PCF8575_LIB_VERSION:\t");
  Serial.println(PCF8575_LIB_VERSION);

  for (int i = 0; i < 3; i++)
  {
    PCF[i].begin();
  }
}

void loop()
{
  for (int i = 0; i < 3; i++)
  {
    for (uint8_t port = 0; port < 15; port++)
    {
      PCF[i].write(port, 1);
      delay(200);
    }
  }

  for (int i = 0; i < 3; i++)
  {
    for (uint8_t port = 0; port < 15; port++)
    {
      PCF[i].write(port, 0);
      delay(400);
    }
  }

}
```

ear9mrn, 2021-12-14:
Thanks Rob, thats excellent. Works perfectly.... I had to make one slight adjustment for my use case, I made the initial declarations "const".

`const PCF8575   A(0x36);
const PCF8575   B(0x37);`

Out of interest. There a couple of other PCF8575 libraries out there and some say you need to set whether a pin is an input or output. I am no expert and just wondering if there a technical reason why this appears not to be required in the case of your library? 

Thanks again,

Pete.

RobTillaart, 2021-12-14:
Thanks for testing, I will add the example to the library.

The answer is rather technical and I'm no electric engineer. So the following is how I recall it (which might be incorrect)
The PCF has no DDR (data direction register)  but a tristate output which can be read back.
By writing a 0 to a pin you switch on a sort of pull down. you will measure 0 volts.
By writing a 1 to a pin you switch on a sort of pull up. you will measure 5 volts (or 3V3 whatever)
However, when writing a 1 the pin becomes HIGH (as above) but when connected to ground it becomes zero. 
This way it is used as an input. 

Read the datasheet (no worries if you not understand all details immediately) to learn how such chip works.

A good book for X-mas: https://www.amazon.co.uk/Practical-Electronics-Inventors-Fourth-Scherz/dp/1259587541/

RobTillaart, 2021-12-14:
FYI Example added to master branch, will be automatically in next release.

ear9mrn, 2021-12-14:
Thanks Rob. Book looks interesting, I will let Santa know...!  I've being playing with the interrupt and i needed to determine which pin initiated the interrupt and I think this may be a good example as well.

after initialising but before setting up the interrupt. 

```uint16_t last_state;last_state = PCF.read16()```

then within the interrupt function.

```uint16_t new_state = PCF.read16();
uint16_t wbit = ( log(new_state -  last_state) / log(2) );
Serial.printf(" %d pin changed....\n", wbit );
last_state = new_state;```

RobTillaart, 2021-12-15:
The PCF.read16 function is way too long to be used in an interrupt routine. You should just set a flag and handle the rad and compare in your main code. 

Note that when you compare the mask it is possible that more than one line has toggled. So you might need to handle at least 1 and maybe 2,3.... 16. 
(Depends on system of course)

ear9mrn, 2021-12-15:
> Thanks Rob. Book looks interesting, I will let Santa know...! I've being playing with the interrupt and i needed to determine which pin initiated the interrupt and I think this may be a good example [...]

One issue i discovered is with my use case during testing is I often got a zero difference! I assume it is due to some kind of contact bounce effect.

my solution! You will probably tell me I am a bit mad to have a (potentially) infinite while loop inside an interrupt!
`
uint16_t diff = new_state ^ last_state;

while (diff == 0){
             new_state = PCF.read16();  
             diff = new_state ^ last_state;
        }

`

RobTillaart, 2021-12-15:
> You will probably tell me I am a bit mad to have a (potentially) infinite while loop inside an interrupt!

😀

From your post I assume that you only have one line changing at the time (sort of keypad / keyboard)?
instead of (expensive) math there is a built in function CTZ  in most compilers you might use.

```cpp
  uint16_t new_state = PCF.read16()
  uint16_t wbit =  __builtin_ctz(a) + 1;                 // +1 to get 1 based counting.
  Serial.printf(" %d pin changed....\n", wbit );
```

endless loop
An other method might be to allow only a new interrupt after e.g. 10 millis() and only if a line is set HIGH.
```cpp
void ISR()
{
  static uint32_t lastIRQ = 0;
  if (millis() - lastIRQ < 100) return;
  lastIRQ = millis();

  uint16_t new_state = PCF.read16();
  if (new_state == 0) return;              // interrupt is caused by a release of the line.
  uint16_t wbit =  __builtin_ctz(a) + 1;                 // +1 to get 1 based counting.
  Serial.printf(" %d pin changed....\n", wbit );
  
  // remaining of the IRQ
}
```

just some thoughts

ear9mrn, 2021-12-17:
Hey Rob,

I have been doing a bit more playing. Sorry if this offends, but I changed how I wanted to implement this library for my use case and I ended up including the declaration inside a struct. I could not get it to work as you advised with the method of using a dummy address then modifying the address.

This is what I came up with.

updated the PCF8575.h with the following additional constructor:

```
PCF8575::PCF8575(TwoWire *wire)
{
  _address    = 0x30; 
  _wire       = wire;
  _dataIn     = 0;
  _dataOut    = 0xFFFF;
  _buttonMask = 0xFFFF;
  _error      = PCF8575_OK;
}
```

added this to PCF8575.cpp:

```
PCF8575(TwoWire *wire = &Wire);
```
Then I was able to place inside my struct:

```
PCF8575   PCF;
```

Then updated the address and called the begin function when required. It appears to work ok for me... Is this a sensible thing to do or have I completely broken your library?

Thanks,

Pete.

RobTillaart, 2021-12-17:
Hi Pete,

It is common practice to have the default parameters ** = &Wire ** in the .h file as that file describes the API / interface.

So you should add in the .h file
```cpp
PCF8575(TwoWire *wire = &Wire);
```

and in the .cpp file 
```cpp
PCF8575::PCF8575(TwoWire *wire)
{
  PCF8575(0x30, wire);   // reuse other constructor to minimize maintenance
}
```

or

.h
```cpp
PCF8575();
```

.cpp
```cpp
PCF8575::PCF8575()
{
  PCF8575(0x30,  &Wire);
}
```

Give that a try.

ear9mrn, 2021-12-17:
Thanks Rob. Tried both methods above but unfortunately I received seg faults for both.

RobTillaart, 2021-12-17:
Will dive into it tomorrow or sunday.
Can you post a minimal sketch that fails and show the construct you try to make?

RobTillaart, 2021-12-18:
Please try the following:
give the address in the existing constructor in the .h file a default value.

```cpp
  // deviceAddress base = 0x20 + depends on address bits
  PCF8575(const uint8_t deviceAddress = 0x20, TwoWire *wire = &Wire);   ///  add = 0x20
```

(Have no hardware to test)

ear9mrn, 2021-12-18:
Confirmed, works perfectly, thanks. Having some issues with i2c bus hanging/locking up but I don't think that is anything to do with the PCF8575 or your library.

RobTillaart, 2021-12-18:
> Confirmed, works perfectly, thanks. Having some issues with i2c bus hanging/locking up but I don't think that is anything to do with the PCF8575 or your library.

Check wires, connections and pull up resistors.  8 out of 10 is hardware. As you have multiple devices on the bus the total capacitance of the wires grows. Reducing the PU resistor might improve communication.
Do you have a scope to see the signals?
If the wires are longer you might need to use shielded wires. I had an I2C display that lost chars by flipping bits due to EMC.
Took days to diagnose and 30 minutes to solve.

BTW I2C clockspeed could be an issue if set above 100KHz but 400 KHz should be doable.

RobTillaart, 2021-12-18:
BTW, I'm going to add that default value to the constructor. It does not break the interface, only allows some extra.

ear9mrn, 2021-12-21:
Thanks for adding that to the the library, very helpful. Been working away trying to get things stable for my use case. Turns out it has nothing to do with the stability of the I2C lines. It is when interrupts are involved and then reading the pins states. Having delays for debounce and/or fast switching is key. I am guessing the PCF8575 does not like a read request too often!  

My final "working" example (see below) that seem pretty stable (based on your example). I added your suggestion with respect to the reading of the bytes using the builtin function. I also added a bit manipulation to determine which bit/pin changed. There is a gotcha here though, which took me a while to figure out. If for some reason there appears to be no change of the pin stats the value it comes up with is 255! I am working with an ESP8266 and so the interrupt function needs to be placed in ICACHE_RAM_ATTR hence the #if/#endif... Also this is only valid for when a single pin changes!

```
#include "PCF8575.h"

PCF8575 PCF(0x20);

const int IRQPIN = 5;
volatile bool flag = false;
uint16_t x, last;
uint32_t elapse;
uint8_t wbit, sda = 10, scl = 9;

#if defined (ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR  void pcf_irq()
#else
  void pcf_irq()
#endif
{
   if ( millis() - elapse > 10 ) {
      flag = true;     
   }   
   elapse = millis();
}

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  delay(random(1000,3000));
  
  Serial.println(__FILE__);
  Serial.print("PCF8575_LIB_VERSION: ");
  Serial.println(PCF8575_LIB_VERSION);

  PCF.begin(sda, scl);
  
  elapse = millis();
  last = PCF.read16();
  
  pinMode(IRQPIN,INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(IRQPIN), pcf_irq, FALLING);
  
}

void loop()
{
  
  if (flag)
  {
    flag = false;
    delay(20);
    
    x = PCF.read16();
    wbit = __builtin_ctz(last ^ x);
    /* Warning....
     * Pin number should be <=15
     * A pin number of 255 means no change 
     * identified in pin states due to 
     * bounce or fast switching?
     */
    Serial.print(x, BIN);
    Serial.print(" Pin: " + String(wbit) );
    Serial.println(" val: " + String(1 & (x >> wbit) ) );
    
    last = x;
    elapse = millis();
    
  }
 
}

```

RobTillaart, 2021-12-22:
Some remarks
- you have a debounce in the IRQ routine and a delay in loop. The second should not be needed. It is the responsibility of the IRQ to guard this debouncing time.  Also keeps your main loop reactive !
- **elapse = millis()** in loop() seems another attempt to throttle the interrupts, should not be needed, see previous.

```cpp
#include "PCF8575.h"

PCF8575 PCF(0x20);

const int IRQPIN = 5;
volatile bool flag = false;
uint16_t x, last;
uint32_t elapse;
uint8_t wbit, sda = 10, scl = 9;

#if defined (ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR  void pcf_irq()
#else
  void pcf_irq()
#endif
{
   if ( millis() - elapse > 20 ) {   ///  <<<<<<<<<<<  note longer bounce period
      flag = true;     
   }   
   elapse = millis();
}

void setup()
{
  while(!Serial);
  Serial.begin(115200);
  delay (random(1000, 3000));
  
  Serial.println(__FILE__);
  Serial.print("PCF8575_LIB_VERSION: ");
  Serial.println(PCF8575_LIB_VERSION);

  PCF.begin(sda, scl);
  
  elapse = millis();
  last = PCF.read16();
  
  pinMode(IRQPIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(IRQPIN), pcf_irq, FALLING);
}

void loop()
{
  if (flag)
  {
    flag = false;
    x = PCF.read16();
    wbit = __builtin_ctz(last ^ x);
    /* Warning....
     * Pin number should be <=15
     * A pin number of 255 means no change 
     * identified in pin states due to 
     * bounce or fast switching?
     */
    Serial.print(x, BIN);
    Serial.print(" Pin: " + String(wbit) );
    Serial.println(" val: " + String(1 & (x >> wbit) ) );
    last = x;
  }
 
}
```

RobTillaart, 2021-12-23:
fyi   0.1.4 released - please confirm nothing broke

ear9mrn, 2021-12-26:
Nothing broke as far as I can tell.

RobTillaart, 2021-12-27:
thanks, that means you can close the issue unless there are related questions.

closed 2021-12-27 by ear9mrn