RobTillaart/PCF8575 #16: Read16() function - strange values are readout from PCF8575

acp-usrmbu, 2022-04-07:
Hello Rob,
many thanks for your email - hope you'll be OK soon.

So my problem is that I 'm not able to read out right value from PCF8575 chip. You can see the ino file which I used for testing... I added only printout the value "x" on screen... the value are strange - random..??
I don't know I do some mistake or what is wrong.... I use an arduino MEGA and a blue PCF8575 module from alliexpress.
Connection: GND, 5V, SDA, SCL. 
There is 4V on the output of the voltage regulator (on PCF8575 module).

It is really strange.
Many thanks for your help :)
Martin

[image]

[image]

RobTillaart, 2022-04-07:
Can you post a close up photo of the PCF?
Both sides please

acp-usrmbu, 2022-04-07:
Dear Rob :)
It seems you opened my eyes :)
I though that after writing some "valueX" by function .write16 than by the next function read16 I readout this number "valueX" back (the value from internal flip-flop).
Now I understand that function "read/read16" reads actual state on the pins of the PCF :-D I'm stupid :) I had to use function valueOut :)

I'm very sorry to bother you....
Many thanks for your help.....
Martin

RobTillaart, 2022-04-07:
If there are no questions left you may clise the issue.

acp-usrmbu, 2022-04-07:
Rob,
I very appreciate your help :)
(it is interesting idea to do such a way I2C memory :) )

Only one next think.

I connected a LED with resistor on one pin. 

P00 ->      --|<|--Resistro ---- +5V   (Led lights OK with "L" state on pin )
but the light doesn't light in this case:
P00->      --|>|-- Resistor ----GND (LED should light with "H" on pin but it doesn't )

It seems the driver doesn't work well.... I tried to bypass the regulator .... I have on chip 5V now... LED doesn't still light... strange

Martin

acp-usrmbu, 2022-04-07:
When I write 0xFFFF there is a 0V on each pin with respect to the GND....
Many thanks
Martin

RobTillaart, 2022-04-07:
0.11V is way too low.
Can you post photo of the board ?
Both sides.

RobTillaart, 2022-04-07:
Do you have a link of the datasheet?
And a link where you bought the PCF?

acp-usrmbu, 2022-04-07:
Rob,
here are photos of my PCF module - both sides.
Datasheet of the PCF and link to sellers page is enclosed too.

Many thanks
Martin

[image]
[image]

Link to seller: 
https://www.aliexpress.com/item/4001273616648.html?spm=a2g0o.productlist.0.0.7fb84600Gck6Wv&algo_pvid=15b745e9-9e4c-40ef-afee-b6b8f7d26a24&algo_exp_id=15b745e9-9e4c-40ef-afee-b6b8f7d26a24-0&pdp_ext_f=%7B%22sku_id%22%3A%2210000015561464537%22%7D&pdp_pi=-1%3B41.07%3B-1%3B-1%40salePrice%3BCZK%3Bsearch-mainSearch

[pcf8575.pdf](https://github.com/RobTillaart/PCF8575/files/8444337/pcf8575.pdf)

RobTillaart, 2022-04-07:
You should check the PCB if all soldering is done well.
Near p05 and p06 there might be an unintended connection.

RobTillaart, 2022-04-07:
From the schema it is clear that the device will operate on 3V3. Not on 5V. That defines what output levels to expect.

acp-usrmbu, 2022-04-09:
> Komu: RobTillaart/PCF8575 ***@***.***>
> readout from PCF8575 (Issue #16)
> "

Dear Rob,

it seems me that there won't be 5V on the output without a load.... the port
has current source from VCC.

So close teh issue please.

Have a nice day

BR

MArtin

---------- Původní e-mail ----------

(https://github.com/RobTillaart/PCF8575/issues/16#issuecomment-1091989983), 
"

closed 2022-04-09 by RobTillaart