RobTillaart/PCF8575 #18: Mixed Inputs and Output pins

IKedros, 2022-05-12:
Hi! First of all amazing driver!

I have a question! In on of the PCF8575 IC I have mixed inputs and outputs. The first 8 (IO0 to IO7) pins are inputs and the last 8 (IO8 to IO15) pins are outputs. How I can define that. I was trying to use the setButtonMask() function but without any luck.

Any help here?

RobTillaart, 2022-05-12:
Thanks for the question, 
Rather busy so might take a few days

RobTillaart, 2022-05-14:
You do not have to set a data direction register with these PCF857x devices. 
You can do a 
```cpp
uint16_t  inputs = PCF.read16() & 0x00FF;. // Select lower 8 bits / lines
uint16_t outputs = b10011101;. // Set some bits high and some low
PCF.write16(outputs << 8);. // Shift them to upper bits.
```
Should do the trick.

IKedros, 2022-05-14:
I was hoping for a direction register but I can see on the datasheet that there isn’t one! 

I will try the code you send me, hopefully it will solve the problem. 
It seems that if I have mixed inputs and outputs on one IC the interrupt pin is always firing! And I can’t solve that somehow…  
at least for now

RobTillaart, 2022-05-14:
You could keep the last / previous value of read16() and compare with current one. 
 mask the output pins and you would see which input triggered the interrupt.

IKedros, 2022-05-15:
The problem is that somehow I cannot reset the interrupt! Since it goes active...  then I can't set it again to the inactive state! And I don't know why is doing that!

IKedros, 2022-05-15:
Ok, it seems that is working now... a read from all pins seems to solve the problem!

RobTillaart, 2022-05-15:
Ok, if solved you may close the issue.

closed 2022-05-15 by IKedros