RobTillaart/PCF8575 #21: frequent reads from a pcf8575

sgrizzi, 2022-07-28:
I need to poll several inputs to check if certain events take place and I use bursts of read() requests.
A read() from the PCF8575 takes ca. 300uS, but the main loop of the esp32 may run faster than this, which could result in a read requests  being issued to the 8575 before the previous one has been completed.
Are requests queued? or is there any "busy" flag which can be used? or I just need to adequately delay the read requests?

RobTillaart, 2022-07-28:
There is no provision for multithreading

RobTillaart, 2022-07-28:
You might need to use semaphores to control the access to the I2C bus.
Note: this problem is larger than just the PCF8575 library, other devices on the same bus will/could be affected too.

sgrizzi, 2022-07-28:
OK, very clear, and it is definitely not related only to the pcf8575...
Thanks!

closed 2022-07-28 by sgrizzi