RobTillaart/PCF8575 #22: tone.h ezBuzzer.h ?

tripod2, 2022-08-21:
not found how to use tone for music
since most of people able to write music are using tone or better nodelay ezBuzzer.h what seem to also include tone
how can i translate it for using multiple buzzers throught pcf8575 ?

RobTillaart, 2022-08-21:
Thanks for this question
I am not familiar with ezBuzzer.
Can you provide a link?

RobTillaart, 2022-08-21:
Found - https://github.com/ArduinoGetStarted/buzzer 

### Quick analysis

Three problems:
1. This library uses the **tone()** and **notone()** functions for the music. 
These functions can only control the hardware pins of the processor, not a PCF8575.
A rewrite of the tone library is needed but even then there is the 2nd problem.
2. To produce a note, the tone library switches on/off a pin at the frequency of  the note. very fast.
The PCF8575 uses I2C which has substantial overhead compared to "local" pins.
(the PCF8575 needs to send its address with every byte send, == substantial overhead)
3. The on off frequency of multiple tones will need many updates so there is a limit of what is possible.
The nr of updates per second are max the sum of the individual max frequencies per channel, and at least equal to the max frequency used.
With 16 channels and 4000Hz max per channel this can be as high as ~65000 updates per second.

### Conclusion (so far)

With a PCF8575 it will not work or at least be difficult to get timing right - (but please prove me wrong here)
You need to rewrite the Tone library

### Idea try 8 channels first

A better chip to use would be the MCP23S08 - a 8 bit SPI based IO - https://github.com/RobTillaart/MCP23S08
This chip can communicate much faster and does not need an address byte so less overhead per update. 
Updating the 8 IO lines can be done in less than 5 micros (UNO)

You should declare a 8 bit variable that represents the state of all 8 buzzers.
If a note needs to switch from on to off or off to on you update this variable and blast it to the MCP23S08
You still need to rewrite the tone library.

### Better idea?

Use an array of AD9850 function generators, so you do not need to generate the frequencies yourself.
Drawback, bigger budget needed.
Advantage there is already thought about a multidevice setup - https://github.com/RobTillaart/AD985X

tripod2, 2022-08-23:
Buzzer passif universel 12805 AC / 2KHz 3V 5V 16 ohms = ok pass
€ 13,39 / lot (5 unités) MCP23S08-E SO no pcb version = fail
€ 13,11 AD9850 for a single Buzzer passif 12805 € 3,18 / lot (50 unités) = fail
 rewrite the tone library = far more easy to rewrite the [PCF8575]
thx for the advice
can you confirm the pcf8575 outputs are grounds ?
i guess i have to add a npn on each output if i want controll a device with common ground like h bridge 
so is a model you can advice ?

RobTillaart, 2022-08-23:
For exact specification you must read the datasheet yourself. A GND in a chip be switched e.g by a mosfet or a pull down, with slightly different behaviour especially timing.
It is important to learn to read these data sheets as it also helps to understand the code in libraries.

It is always better to add a transistor behind these kind of IO chips to be able to control some more serious load.

tripod2, 2022-08-23:
i have a limited understanding of how the electronic work so its mandatory for me and i guess for all the beginner that some expert can read the datasheet and resume only the meaningfull data for arduino programmation
thx to you alls
i want just powerup a buzzer or a h bridge trought it nd i have a box of common transistor of many values but im unable to decide what is the most optimised for this task and what is under or over specification
can you advice and give some standard exemple of a good transtior to solder at the pins ?
also wish to use a 2 watt led or speaker another advice ?

RobTillaart, 2022-08-23:
The Arduino forum is a better place and sites like stackexchange to learn about electronics. I am a software guy and on those fora there are real electronic guru's with far more knowledge than I have.

Best advice is to get a copy of Practical Electronics for Inventors. That book will learn you everything you need about electronics and way more. And yes it will cost serious time to master it. 

Learning to read a datasheet is important as ALL information in a datasheet is important. Libraries are convenient to get started, but if you need more one should start digging in the datasheet, compare, prototype, experiment, start over etc.

The questions you have are out of scope for this library and its functionality. Arduino forum section project guidance might be the place to ask.

tripod2, 2022-08-24:
be sure than if i ask here its because i not found any tutorial elsewhere and since its you the source it you the well placed for aNSWERING
so i used the p0 as ground and the vcc as positive for a led and its just not work more than the reverse
what i made wrong ? using exemple only
you can check search yourself their is no pinout diagram on the net
maybe this pcb is not interseting the mass for some reason
i can not be sure if its defective or functionnal
i spended a whole day to try all of the 6 library compatible ith arduino ide and nothing happended

RobTillaart, 2022-08-24:
The **PCF8575** is indeed far less used.
However there should be plenty diagrams of the 8 channel **PCF8574** available which is far more used. And as far as I recall these are quite similar except for number of channels.

To control a LED you must at least use a current limiting resistor in series with the LED. Furthermore the polarity of the LED should be correct. You need to look up th current the LED can handle, measure the voltage of VCC to calculate the resistor with OHMs law. Then you should check the datasheet of th **PCF8575** if it can provide that current needed. If not you need to use a MOSFET or a transistor in between. Or a darlington or an H bridge. 

Schemas for how to connect a transistor to an IO are widely available. These are also used when connecting a relay to an Arduino or any other processor. 

The book I mentioned describes all of this.

RobTillaart, 2022-08-24:
To analyse your specific problem the Arduino forum is a far more appropriate place as there are electronic experts to explain the electronic details far better than I can.

RobTillaart, 2022-08-24:
> you can check search yourself their is no pinout diagram on the net

GOOGLE: Arduino PCF8575 tutorial  (search imageas) 
https://www.mischianti.org/2019/07/22/pcf8575-i2c-16-bit-digital-i-o-expander/

and dozens more.

tripod2, 2022-08-24:
lol i seen that and if you considere the pcf8575 as a led controller what im not even able to complete usaing those library

RobTillaart, 2022-08-25:
Maybe your chip is broken?
Does the processor see the chip on the I2C bus?
Check signals with a scope or a digital multimeter.

tripod2, 2022-08-25:
the ic2 is ok and i also can read 
but the value look not straight when i set all to low i get all zero but when i set  all high sequentially or by instruction their is a random mix of 1 and 0

RobTillaart, 2022-08-25:
Can you read both HIGH and LOW values?

> but when i set all high sequentially or by instruction their is a random mix of 1 and 0

That is not how it should be. 
- Which example sketch do you use?
- Can you post a schema of your setup?
- Do you have a multimeter?

tripod2, 2022-08-26:
i put a led on the p0 pin and a voltmeter on the pin p1
knowing the output is ground i reversed them
i tried all the pcf8575 what is about 4
everytime i get no output 
i used the print (pcf.read16 ); for testing all instruction and only the unselect all showed straight 000000000000 all the other showed a mix of 0 and 1

RobTillaart, 2022-08-26:
Can you post a schema of your setup?

RobTillaart, 2022-09-09:
As there is no response I assume the issue is solved.