RobTillaart/PCF8575 #27: Nothing will compile!

jsmwrench, 2022-12-31:
Hello!

I've been trying to get the PCF8575 digital multiplexer module to work on an ESP8266 DevKit 1 and I can't get any of the examples in your code folder to even compile in the Arduino IDE.  I deleted the pertinent libraries from my files and installed new, freshly downloaded versions of both this library and the Wire library (using the "Add ZIP library" function) and I keep getting the same errors.

The error that shows up first is this one:  
`c:\Users\jsmwr\Documents\Arduino\libraries\PCF8575\PCF8575.cpp: In member function 'bool PCF8575::begin(int, int, uint16_t)':
c:\Users\jsmwr\Documents\Arduino\libraries\PCF8575\PCF8575.cpp:33:35: error: no matching function for call to 'TwoWire::begin(int&, int&)'
   33 |     _wire->begin(dataPin, clockPin);`

I've had this same error using other libraries for PCF8575, so I'm not sure what's going on.

Thank you for your time,
Ron.

RobTillaart, 2022-12-31:
Thanks for this issue. 
I'll try to look into this asap. Seen similar error before

jsmwrench, 2022-12-31:
I appreciate the quick response.  Thank you kindly.

jsmwrench, 2022-12-31:
The compiler presents quite a few "notes" also.  Would you like me to post those as well?

jsmwrench, 2022-12-31:
By the way, the Wire2 example gave me different error:
`C:\Users\jsmwr\Documents\Arduino\libraries\PCF8575\examples\PCF8575_Wire2\PCF8575_Wire2.ino:11:20: error: 'Wire2' was not declared in this scope; did you mean 'Wire'?
   11 | PCF8575 PCF(0x39, &Wire2);
      |                    ^~~~~`

RobTillaart, 2022-12-31:
https://github.com/RobTillaart/PCF8575/actions/runs/3517394399/jobs/5895114738
shows that the ESP32 does compile things,

But lets try to compile **PCF8575_test.ino** for ESP32 here (IDE 1.8.19)

RobTillaart, 2022-12-31:
OK, ESP32 compiles with known "warnings". That is a good sign.

As the ESP8266 is not tested in the github build, and I do not have an ESP8266 nearby (if at all) so I can only do some compile tests and hope to find the cause of the error.

jsmwrench, 2022-12-31:
It's interesting that the ESP32 compiles but not the ESP8266.

RobTillaart, 2022-12-31:
> It's interesting that the ESP32 compiles but not the ESP8266.

The ESP8266 at least the generic one does compile too
```
Sketch uses 269881 bytes (28%) of program storage space. Maximum is 958448 bytes.
Global variables use 28540 bytes (34%) of dynamic memory, leaving 53380 bytes for local variables. Maximum is 81920 bytes.
```

jsmwrench, 2022-12-31:
Hmm... Okay.  I'll try that.

RobTillaart, 2022-12-31:
I will have a look into the ESP8266 Wire library interface. (might take 15 minutes ...

RobTillaart, 2022-12-31:
I have ESP8266 version 3.0.2

Snippet
```cpp
class TwoWire : public Stream
{
private:
    static uint8_t rxBuffer[];
    static uint8_t rxBufferIndex;
    static uint8_t rxBufferLength;

    static uint8_t txAddress;
    static uint8_t txBuffer[];
    static uint8_t txBufferIndex;
    static uint8_t txBufferLength;

    static uint8_t transmitting;
    static void (*user_onRequest)(void);
    static void (*user_onReceive)(size_t);
    static void onRequestService(void);
    static void onReceiveService(uint8_t*, size_t);
public:
    TwoWire();
    void begin(int sda, int scl);             <<<<<<<<<<<<<
    void begin(int sda, int scl, uint8_t address);  
```

that matches the line ```    _wire->begin(dataPin, clockPin);````

[image]

jsmwrench, 2022-12-31:
Well, that's what was confusing to me, because (even though I don't really know how to read/write libraries) it seemed like it should work as it's written.

jsmwrench, 2022-12-31:
I tried using the generic 8266 and still got the same compile error.  I don't know what's going on.

RobTillaart, 2022-12-31:
Can you check in the IDE under FILE -> PREFERENCES if you have this board manager?

http://arduino.esp8266.com/stable/package_esp8266com_index.json

jsmwrench, 2022-12-31:
Yes, it's in the list.
[image]

RobTillaart, 2022-12-31:
OK, that looks good
Can you check in the IDE  TOOLS-> BOARD -> BOARD MANAGER if there are updates pending?

-----

your error message shows line 33 is in error which indicates that at the ESP8266 flag is defined..

Pleas note that the library overwrites the **\_wire** so passing Wire2 as parameter does not work. 

```cpp
#if defined (ESP8266) || defined(ESP32)
bool PCF8575::begin(int dataPin, int clockPin, uint16_t value)
{
  _wire      = &Wire;                  <<<<<<<<<<<<<<<<<<<<<<<<<<<
  if ((dataPin < 255) && (clockPin < 255))
  {
    _wire->begin(dataPin, clockPin);
  } else {
    _wire->begin();
  }
  if (! isConnected()) return false;
  PCF8575::write16(value);
  return true;
}
#endif
```

RobTillaart, 2022-12-31:
you should see something like this
[image]

jsmwrench, 2022-12-31:
Okay. There are crap-ton of 8266 libraries in there, but I don't see that one.  That must be the issue.  I'm not sure where it went, since I was using 8266's last year with no issue.  Anyway, thank you for your time and I hope you don't feel it was wasted.

I appreciate all that you went through.

RobTillaart, 2022-12-31:
Welcome,

jsmwrench, 2022-12-31:
Sorry to bother.... but where can I get that library you showed?  It's not in my Arduino IDE.  I know it used to be, but it's not there now.

jsmwrench, 2022-12-31:
Never mind.  I thought you were showing me a library, but it was the board.  I do have that one and I still get the same error.

RobTillaart, 2022-12-31:
Think the scenario is 
first remove the folder somewhere under /user/you/appdata/arduino/packages (something like that).
Then remove the link from file-preferences 

Restart the ide (all instances) 
Add the link again to file preferences
And then install board again from boardmanager.

(Laptop is off as here it is almost new year, lots of firework already, party time has started :)

jsmwrench, 2022-12-31:
Happy New Year.... thank you again

jsmwrench, 2023-01-01:
I know you're signed off, but I wanted to let you know that fixed it all....including the errors I was having with the PCF8574.  I'm not sure what went wrong and/or where, but the issue is resolved.  Thank you again.

closed 2023-01-01 by jsmwrench

RobTillaart, 2023-01-01:
Happy new year!
Good to hear the issue is solved, that is a good start of the new year!