RobTillaart/PCF8575 #29: Support for Teensy 4.0 multiple i2c

DennisSchm, 2023-07-10:
Hello and thank you for this great library. 

I'd like to why your library does not allow for constructors passing the **Wire** address when **TEENSYDUINO** is defined. 
I am using the teensy 4.0 which has more than one wire but without modification of the library I could only use the default wire. This should be a fairly easy change to do if I don't miss something. 

Thank you for your time, 
Dennis

RobTillaart, 2023-07-10:
Good morning, 
The reason is that my library development process is evolutionary. They grow on request if request is generic enough. Some requests make sense for multiple libraries so if time permits I add them to those that apply. This request makes sense for at least PCF8574 and possibly all I2C based libraries.

I will dive into this asap. As I do not have all boards I hope you can do the testing.

RobTillaart, 2023-07-10:
@DennisSchm 

Installed **TEENSYDUINO** in my 1.8.19 IDE and **PCF8575_Wire2.ino** test sketch compiles.
Cannot test it as I have no teensy 4.0

RobTillaart, 2023-07-10:
Wrong library error => continues in #30

closed 2023-07-10 by RobTillaart