RobTillaart/PCF8575 #30: examples including PCF8575.h instead of PCF8575_library.h

DennisSchm, 2023-07-10:
Hello and thank you for your work, 

while reading through the library code I noticed that you introduced include guards in a separate file called **PCF8575_library.h**.
Later the example files omit those guards by including **PCF8575.h**. This could also happen to other users not having a look at the library at all or following those examples. I think it would be no problem to include the guards directly into **PCF8575.h** bypassing this problem completely. 

Thank you for your time, 
Dennis

RobTillaart, 2023-07-10:
Will look into this one asap.

RobTillaart, 2023-07-10:
Labeled as bug for now.

RobTillaart, 2023-07-10:
@DennisSchm 

Hi Dennis,
Did not find any library of me mentioning **PCF8575_library.h** (except this issue)
Can you post a link where you see this line?

Maybe an old version had it  - I do not recall - so update to the latest version 0.1.8 should solve this.
If you downloaded a fork, then you should create an issue at that fork.

DennisSchm, 2023-07-10:
You are right, thats my fault. I confused your library with [PCF8575 library by Renzo Mischianti](https://github.com/xreef/PCF8575_library) as I was to believe PlatformIO recommends yours first and I was editing it, while I was not. Please excuse the inconvenience. I believe the same thing applies to: #29 . I think both issues can be closed.
If I still can be of any service testing for the teensy, I will gladly do so.

RobTillaart, 2023-07-10:
Another library could also be a cause.

Worked about two hours on both issues, searching, installing teensy etc.
Confirmed that the library compiles for **TEENSYDUINO**, 
If you can confirm this with a hardware test time is well spend.

DennisSchm, 2023-07-10:
I will do so today and get back at you when I'm done.

RobTillaart, 2023-07-11:
Super, that is very well tested. 
Will look into improving platform-io support later.
Thanks a lot, really appreciated!

RobTillaart, 2023-07-18:
As this issue in itself is solved I close it for now.

closed 2023-07-18 by RobTillaart