RobTillaart/PCF8575 #34: How use address 0x39

pierrotm777, 2023-12-29:
Hello,

Thanks for your nice library :-)

I try to use a pcf8575 for 16 inputs.
All the docs i have found say that the address are from 0x20 to 0x27 and in fact,
i find my default address at 0x20.
But your code use often the address 0x38 or 0x39.
Ideally, i should like to use the address 0x3C to 0x3F if possible (i think it's possible with the pcf8575A only).
How do you config the pin A0 to A2 for have these address please.

Thanks for your help,

Pierre

pierrotm777, 2023-12-29:
And if i test the setAddress(0x0F), getAddress() return me well 0x3F but isConnected return me false.
My code:
  PCF.setAddress(0x3F);
  delay(2000);
  PCF.begin();
  uint8_t getAdd = PCF.getAddress();
  
  Serial.println(getAdd,HEX);

RobTillaart, 2023-12-29:
Thanks for your question,

The examples are reused from the PCF8574 which has addresses in 0x20..0x27 range and the PCF8574A which uses addresses in the 0x38..0x3F range. A quick look did not show no PCF8575A series, so the examples need to be adjusted. 

I will patch the examples so they use a valid address, to remove this point of confusion. 
Sorry for the inconvenience.

RobTillaart, 2023-12-29:
**bool setAddress(const uint8_t deviceAddress)** sets the device address after construction. Can be used to switch between PCF8575 modules runtime. Note this corrupts internal buffered values, so one might need to call read16() and/or write16(). **Returns true if address can be found on I2C bus.**

It does not force an address, think of it as select the device with address xyz.
And as stated above it has some (nasty) side effects.

RobTillaart, 2023-12-29:
Your question popped the concept in my mind of an "I2C-address change chip". 
Imagine a chip like a multiplexer that can change the address in an I2C call runtime. 
Never heard of it, so need to search e.g. NXP or TI site.
If you have a board with two I2C ports it could be created. 
(OK enough daydreaming, lets fix the bug)

pierrotm777, 2023-12-29:
Thanks for these answers :-)

RobTillaart, 2023-12-29:
I see that the readme states it correctly.

RobTillaart, 2023-12-29:
Will add a check in **setAddress()** to return false if address is out of range.

RobTillaart, 2023-12-29:
Develop branch created, 
If you have time can you please verify it?

pierrotm777, 2023-12-29:
Thanks, it's ok now for me.
Do you have done a library for the AW9523 witch use address from 0x58 to 0x5B ?

RobTillaart, 2023-12-29:
What is the AW9523 as I am not familiar with it.
Do you have a link to the datasheet?

closed 2023-12-29 by RobTillaart

RobTillaart, 2023-12-29:
Found 

- https://cdn-shop.adafruit.com/product-files/4886/AW9523+English+Datasheet.pdf
- https://github.com/adafruit/Adafruit_AW9523

pierrotm777, 2023-12-29:
Exact, it's for drive LED but i think it's can bin used as 16 inputs, no ?

RobTillaart, 2023-12-29:
- https://github.com/mrmx/AW9523B  this one too

https://www.adafruit.com/product/4886  states it is 16 IO so input should be an option.

pierrotm777, 2023-12-29:
Thank you RobTillaart :-)

commit 65aaf312 (2023-12-29): Fix #34, correct addresses in examples (#35)

- Fix #34 change addresses in examples.
- add range check to **setAddress()**.
- update readme.md.
- correct minor error in changelog.md.