RobTillaart/PCF8575 #36: One off my board isn't run

pierrotm777, 2023-12-30:
Hello again Rob,

I have a curious issue with one off my two board.
Several boards exist, a red or a blue.
The blue has a level converter from 5v to 3,3v but i use the pin VDD for the 5V.
The red has no level converter and no pull-up. I have added two 10k resistors.
I use this code for check my two board.
[BURC_16Switchs_Test2.zip](https://github.com/RobTillaart/PCF8575/files/13798789/BURC_16Switchs_Test2.zip)
The blue is running very well but not the red.
The isConnect return me OK for the two boards.
Do you know this difference ?
Thanks for your help.

Pierre

RobTillaart, 2023-12-30:
Try lower the pull ups to 4k7 or even 2k2.

RobTillaart, 2023-12-31:
learn all you need to know about i2c and more - https://www.gammon.com.au/i2c

pierrotm777, 2023-12-31:
Hello Rob,
I have replaced the 10k by 2 4.7k with the same issue.
Now, if i add a pull-up on the P0 pin, this pin is usable as input.
Does exist a way for setup the pins as input with pull-up ?

RobTillaart, 2023-12-31:
Not that I know, 
The detailed behaviour of the chip is described in the datasheet, furthermore you need to check the schema of your board to understand how the chip is used. That can be different per board.

RobTillaart, 2024-01-04:
Any progress to report?

pierrotm777, 2024-01-04:
Hello Rob,

Yes, i have found that my pcf8575 red board is a PCF8575C version.
In input mode, it need 16 pull-up and that solve my issue with this component.
Without pull-up, the input are always set to a low state.

Thanks again for your help:-)

And I wish you a Happy New Year 2024.

Pierre

RobTillaart, 2024-01-04:
Q: do you have a link to a datasheet of the PCF8575C ?

closed 2024-01-04 by RobTillaart

pierrotm777, 2024-01-04:
No sorry.
I have just find some links about pcf857c witch need pull-up.

munkerench, 2024-05-30:
Oddly found the data sheet here https://www.nxp.com/docs/en/data-sheet/PCF8575C.pdf

pierrotm777, 2024-05-30:
Thank you for your find

RobTillaart, 2024-05-30:
Added the datasheets PCF8575 and PCF8575C to this repo