RobTillaart/PCF8575 #39: read16() is in Interrupt-Routine not possible

Torsten61, 2024-02-10:
I wanted to cache/latch the current inputs in an array, because of the slow computer program in the Interrupt call.  
Later, the data should be processed in the slow program one after the other (FIFO).

BUT:  Unfortunately, read16() does not run in the IR-routine....

Suggested solutions?

Torsten61, 2024-02-10:
>   Thanks for the question. What board are you using?

Arduino Mega2560 with RS485 and 30 IOs--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

RobTillaart, 2024-02-11:
Same advice as in your previous question.
You need a chip with an input latch to solve this.

Given you have a Mega,  dont you have still some free pins?
If you monitor all input with on board pins you could use a pin change interrupt library. 
Output lines might be moved to PCF8575.
Don't know your requirements so I can only guess if a redesign would help.

RobTillaart, 2024-02-11:
This library has no solution for it as far as I know.

Torsten61, 2024-02-12:
So I have to use another HW or optimize the prog-code and use IQR-FLags - Thanks for help!

closed 2024-02-12 by Torsten61