RobTillaart/PCF8575 #40: wrong return on begin

DavidNidam, 2024-04-09:
Hey

Call begin for the first time when no using interrupt and writeMode & readMode == 0 return fail because transmissionStatus == 4.

Reproduce:
1. run ESP8266
2. define PCF8575 pcf8575(0x20);
3. call:     if (pcf8575.begin())
        Serial.println(F("OK"));
    else
        Serial.println(F("Failed"));
4. wrong behavior... always return fail

RobTillaart, 2024-04-09:
Thanks for the issue, 
Will look into it later this week (busy)

RobTillaart, 2024-04-09:
transmissionStatus == 4

Thanks for this analysis, need to check the details if the I2C implementation of the ESP8266.

RobTillaart, 2024-04-09:
@DavidNidam 
### Wire.cpp

```cpp
uint8_t TwoWire::endTransmission(uint8_t sendStop)
{
    int8_t ret     = twi_writeTo(txAddress, txBuffer, txBufferLength, sendStop);   <<<<<<<<<<
    txBufferIndex  = 0;
    txBufferLength = 0;
    transmitting   = 0;
    return ret;
}
```

### core_esp8266_si2c.cpp
```cpp
    uint8_t twi_writeTo(unsigned char address, unsigned char* buf, unsigned int len,
                        unsigned char sendStop)
    {
        return twi.writeTo(address, buf, len, sendStop);   <<<<<<<<<<<<<<<
    }
```

### core_esp8266_si2c.cpp  (again)

```cpp
unsigned char Twi::writeTo(unsigned char address, unsigned char* buf, unsigned int len,
                           unsigned char sendStop)
{
    unsigned int i;
    if (!write_start())
    {
        return 4;  // line busy   <<<<<<<<<<<<<<<<<<<<<<<<<<<<  Here is your 4 
    }
    if (!write_byte(((address << 1) | 0) & 0xFF))
    {
        if (sendStop)
        {
            write_stop();
        }
        return 2;  // received NACK on transmit of address
    }
    for (i = 0; i < len; i++)
    {
        if (!write_byte(buf[i]))
        {
            if (sendStop)
            {
                write_stop();
            }
            return 3;  // received NACK on transmit of data
        }
    }
    if (sendStop)
    {
        write_stop();
    }
    else
    {
        twi_scl_valley();
        // TD-er: Also busywait(twi_dcount) here?
        // busywait(twi_dcount);
    }
    i = 0;
    while (!SDA_READ(twi_sda) && (i++) < 10)
    {
        twi_scl_valley();
        busywait(twi_dcount);
    }
    return 0;
}
```

### core_esp8266_si2c.cpp  (again)

```cpp
bool Twi::write_start(void)
{
    SCL_HIGH(twi_scl);
    SDA_HIGH(twi_sda);
    if (!SDA_READ(twi_sda))
    {
        return false;    <<<<<<<<<<<<<<<<
    }
    busywait(twi_dcount);
    // A high-to-low transition on the SDA line while the SCL is high defines a START condition.
    SDA_LOW(twi_sda);
    busywait(twi_dcount);
    // An additional delay between the SCL line high-to-low transition and setting up the SDA line
    // to prevent a STOP condition execute.
    SCL_LOW(twi_scl);
    busywait(twi_dcount);
    return true;
}
```

So your I2C seems not to be in a ready or start state.
Did you call **Wire.begin()** in your sketch ?

DavidNidam, 2024-04-10:
Hey

 

No need to reproduce… just look at the code…

 

Line 167 -> this->transmissionStatus = 4;

Line 192-> if (writeMode>0 || readMode>0) -> false

Line 211-> if (_usingInterrupt) ->  false

 

Results: transmissionStatus at 4 (never changed)

 

Thus line 220-> return this->isLastTransmissionSuccess(); -> return transmissionStatus==0;

IE begin return false even though everything is OK

 

David

 

I cannot reproduce from that, but I will do a call stack trace

RobTillaart, 2024-04-10:
You refer to line numbers of which file?

DavidNidam, 2024-04-10:
> @DavidNidam <https://github.com/DavidNidam>
>
> Your error comes from the PCF8575 library from xreef.

Yep

That’s what I referred to…

https://github.com/xreef/PCF8575_library/blob/master/PCF8575.cpp#L167

RobTillaart, 2024-04-10:
> Yep That’s what I referred to…

That's not my library, so you need to open an issue at his repository.

RobTillaart, 2024-04-16:
@DavidNidam 
As this is not an issue of my library I close the issue.

closed 2024-04-16 by RobTillaart

DavidNidam, 2024-04-17:
Sure.
Sorry for the mistake

Dady