RobTillaart/PCF8575 #46: PCF8575 -  Chain - No Communication

penguinpages, 2025-04-28:
I am looking for user forum for this board.  First time trying to use this board

I have purchased four of them for a project and so need to address the full 4 x 16

Starting with first board set to x20 (DIP 1-3 ON)  I connect to my ardunio  R4 wifi:

Basic Connection:   
Ardunio SCL -> board (green post but also tried black pin post) SCL
Ardunio SCA -> board (green post but also tried black pin post) SCA'

use "test" code to scan for board:

```
#include "PCF8575.h"

//  adjust addresses if needed
PCF8575 PCF(0x20);

void setup()
{
  //  while(!Serial);  //  uncomment when needed
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("PCF8575_LIB_VERSION:\t");
  Serial.println(PCF8575_LIB_VERSION);
  Serial.println();

  Wire.begin();

  if (!PCF.begin())
  {
    Serial.println("could not initialize...");
  }
  if (!PCF.isConnected())
  {
    Serial.println("=> not connected");
  }
  else
  {
    Serial.println("=> connected!!");
  }
}

void loop()
{
}

```

Make sure "serial out"  baud set to 115200

Output

```
17:33:12.011 -> could not initialize...
17:33:12.011 -> => not connected
09:43:04.167 -> could not initialize...
09:43:04.167 -> => not connected
09:44:25.920 -> could not initialize...
09:44:25.920 -> => not connected
```

RobTillaart, 2025-04-28:
Thanks for reporting the issue.
I will try to come back to it later today.

You can always discuss items on the Arduino forum. There are many volunteers helping people for answers.

RobTillaart, 2025-04-28:
Do you have the address pins of the PCF8575 connected to force the address used?

If you connect four, everyone should have its own unique address. (but that is step 2)

Do you have an I2C scanner to see if the address is correct?

RobTillaart, 2025-04-28:
> And no.. I do not have I2C scanner... but if I can use Arduino or PI V as that role and you have code

you can build your own I2C scanner with this library, or use one of the examples.
- https://github.com/RobTillaart/I2C_SCANNER

RobTillaart, 2025-04-28:
> and I will add to my debug skills / tools

👍

RobTillaart, 2025-04-28:
From the photo part II

You must connect at the ground of the PCF board with the ground of the Arduino board.
This will give them a common reference of the signals.

penguinpages, 2025-04-28:
I did that.  Can I also assume that the pin posts in middle (black standoff with six pins)  of board vs "green lug contacts"  (two for power and two for data) are the same?

RobTillaart, 2025-04-28:
[image]

That seems like an alternative I2C connector. 
I expect it is 1-1 connected with the green screw terminals at the other end.

RobTillaart, 2025-04-28:
Found this link - https://www.iotmug.com/16-channel-i2c-relay-electromagnetic-33v-50v-smart-home-assistant-esp32-esp826

it has some good pictures, lets see if it pops up some documentation.

RobTillaart, 2025-04-28:
[image]
Connector 1

[image]
The jumpers are for the address

[image]
The other connector

RobTillaart, 2025-04-28:
From that site
- For stable operation with 3.3V, an I2C logic level shifter should be used
- Maximum current @5.0V – 1500mA  (power 3amp 5v == OK)

RobTillaart, 2025-04-28:
Do you have the GND of the power supply of the board (WHITE if I am correct) connected to the GND of the Arduino board 

It is not visible on your photo

https://europe1.discourse-cdn.com/arduino/original/4X/1/e/d/1edf8a2ec2988fc14c326e3a6cab5e57b2e68b74.jpeg

penguinpages, 2025-04-28:
Sees you founds similar manual as I did. 

yes.  and moved  all posts over to "black" standoff  to baseline.

[image]

Sorry hand in way.  I was verifying 5vdc on green posts and trying to take picture :)

RobTillaart, 2025-04-28:
OK with 3 switches to ON, I expect the address is 0x27.
if you change that in the code I expect it to work. (or change the jumpers)

RobTillaart, 2025-04-28:
The I2C scanner would just check which addresses it sees.
It would use minimal current, so it should work at 5V which you measure.

> its like I am missing something stupid simple

like swapping 5V and GND?

penguinpages, 2025-04-28:
I posted to forum..  Even tried to power board from direct 5v on arduinio, which I assume is not best practice long term :) 

https://forum.arduino.cc/t/pcf8575-chain-no-communication/1377343/9

No change

RobTillaart, 2025-05-02:
@penguinpages 
I saw on the forum you are able to see the board now if you have a sufficient strong power supply.

The base schematic should be like this
```
      POWER SUPPLY            PCF8575 BOARD                  Arduino R4
   +----------------+      +--------------------+         +-----------------+
   |                |      |                    |         |                 |
   |                |      |                SDA |<------->| SDA             |
   |                |      |                SCL |<--------| SCL             |
   |            + V |------| + V            +V  |         | +V              |
   |            GND |------| GND            GND |---------| GND             |
   |                |      |                    |         |                 |
   +----------------+      +--------------------+         +-----------------+
```

LEFT: A power supply connects to the PCF8575 board (+V , GND) 
RIGHT: The Arduino R4 connects  to the PCF8575 board (SDA, SCL, GND)
The Arduino R4 has its own power supply / could be USB to a PC

Most important is the GND line between PC8575 and Arduino R4 as it gives a common reference for the SDA/SCL signals.

Can you confirm you tested this setup?

penguinpages, 2025-05-03:
Thanks for updates and posting  The diagram helps.    So to power board  5v.  Do you have table of what the matrix is for amps for board count? 

Ex:  
1boards - 5v@1amp
2boards - 5v@1.5amp
3boards - 5v@1.5amp
4boards - 5v@2amp
etc.. up to maximum 8 boards

Second question is the range for connecting common:
All commons support 1-48v @ each contact 4 amp max? etc.. 

My plans are very much more modest for this project   12v  max 1amp connection device  But trying to make sure I don't repeat yak shaving over being out of spec and that creating flaky results.

As always, thanks.

RobTillaart, 2025-05-04:
@penguinpages 

> Do you have table of what the matrix is for amps for board count?

No, should be in datasheet / description of the board. 
As the relays draw the most you might expect have the peak current when they all switch on simultaneously. 
My assumption is that the relays are mechanical (please verify)
if so, the Arduino can send signals to all boards to switch on even before the first board has finished.
So there might be a 

If one board uses 1A at max you just need to multiply that by the number of boards,
Add a safety margin of factor 1.5 or 2 and you will probably on the safe side. 

You should consider to have a power supply per board as these would be smaller and could add to robustness.
However I do not know your requirements if multiple power supplies is an option.

penguinpages, 2025-05-07:
Yes.  This was a factor of   1) Need to use ground from arduino board   2) Need higher amperage power supply on the arduino board.  it was powering on but without means to know for sure.. was not sufficent draw to get it stable.

Thanks for help. Look forward to using boards

RobTillaart, 2025-05-07:
Thanks for the update, I will close the issue.